har-in-air/STM32F411_USB_AUDIO_DAC #3: Difficulty getting Make to compile

Jaron123, 2021-05-05:
Hello,

When trying to build the project I get:
"

make all 
arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DUSE_FULL_ASSERT -DSTM32F401xC  -DDEBUG_FEEDBACK_ENDPOINT  -Iinc -Idrivers/STM32F4xx_HAL_Driver/Inc -Idrivers/STM32F4xx_HAL_Driver/Inc/Legacy -Istm32_usb/Core/Inc -Istm32_usb/Class/AUDIO/Inc -Idrivers/CMSIS/Device/ST/STM32F4xx/Include -Idrivers/CMSIS/Include -Idrivers/BSP -O2 -Wall -fdata-sections -ffunction-sections -MMD -MP -MF"build/main.d" -Wa,-a,-ad,-alms=build/main.lst src/main.c -o build/main.o
c:/st/stm32cubeide_1.6.0/stm32cubeide/plugins/com.st.stm32cube.ide.mcu.externaltools.make.win32_1.5.0.202011040924/tools/bin/sh: arm-none-eabi-gcc: not found
make: *** [Makefile:180: build/main.o] Error 127
"make all" terminated with exit code 2. Build might be incomplete.

"
I'm new to makefiles and compiling with STM32CubeIDE but so this might be inexperience but I'm at a loss of what to do.
Am I missing something? Is the project still ready to build?

har-in-air, 2021-05-05:
In the makefile, since GCC_PATH is not defined for you,  set GCC_PATH to the full windows path to the ST gcc compiler binary. Or you could add the path to the binary to your windows environment variable PATH.

closed 2021-05-06 by har-in-air

Jaron123, 2021-05-06:
Thanks Hair in air!
If anyone has the same problem I copy and pasted the PATH variables (Project-properties-Enviroment) from another project into this one and that seemed to work.

FAUSTINOCRUZELIS, 2022-09-14:
> Thanks Hair in air! If anyone has the same problem I copy and pasted the PATH variables (Project-properties-Enviroment) from another project into this one and that seemed to work.

hello bro, i didn''t undestand how do you solved it 
what software o binaries do you used to compile this on windows (i had to install ubutu and a lot of packets to make it compile jaja)