har-in-air/STM32F411_USB_AUDIO_DAC #5: Problem with compiling

gotrune, 2021-05-14:
Hello, 
I'm on windows with cubeIDE v1.6.1 and I cant compile due to librairies error, i want to try asap if this board is good for me then can you upload a compiled program for the stm32f401CU please?

har-in-air, 2021-05-14:
??!!  What is the library error ? What is "this board" ? You have provided zero information that points to an issue with the repository code that I need to fix.

closed 2021-05-14 by har-in-air

gotrune, 2021-05-14:
I dont have any link for the library's in the project. It's my first time on CubeIDE.
I think there's no issue with your code, i'm a beginner with cube and i dont want to spend more time on its. My test board is a Blackpill STM32F401CCU6, I just want a compiled file please. 
If the project work good for my purpose i'll gop further with the debug....