har-in-air/STM32F411_USB_AUDIO_DAC #6: Help troubleshooting, please

robcazzaro, 2021-05-16:
I'm using a Windows 7 PC connected to a BlackPill STM32F411 with your code running on it. Everything works

From time to time, though. I get the onboard blue LED (PC13) to turn on, flashing rapidly. The code that triggers it is in usb_audio.c, in particular:

```
    // Monitor remaining writable buffer samples with LED
    if (audio_buf_writable_samples < AUDIO_BUF_SAFEZONE_SAMPLES) {
    	BSP_OnboardLED_On();
    	}
    else {
    	BSP_OnboardLED_Off();
    	}
```

I'm trying to understand if that's something I need to worry about and, if so, what can I look at to try and figure out if I did something wrong (I'm compiling on a Windows machine and using a STM32CubeIDE project I created, so there might be some difference with your code)

har-in-air, 2021-05-17:
If the led is flashing on and off regularly and the audio application is not working, it's in the Error_Handler in main.c - which means a fatal fault. I haven't seen that yet, you can try putting a breakpoint in the entry and trace the call leading to the fault. 

If the blue led comes on while no audio is streaming, and then immediately goes off when you start streaming audio, no problem.

closed 2021-05-17 by har-in-air

robcazzaro, 2021-05-17:
Thanks for your reply. I should have mentioned, but I already had a breakpoint in the Error_Handler function, and that never gets triggered.

In my case the blue led turns on and off while streaming. The BSP_OnboardLED_On(0 and BSP_OnboardLED_Off() functions sometimes get called multiple times while streaming, so I can see the LED flickering. 

Am I right in understanding that the LED flickering while streaming might indicate a buffer underrun? I wonder if it's a problem with the USB bus on my Windows machine being overloaded and maxing out, since it seems to happen more frequently when the PC is overloaded

har-in-air, 2021-05-18:
Thats possible. Do you have heavy data transfers on USB e.g. external hard drives - that could cause an issue with available bus bandwidth for isochronous transfers. I have two USB hubs attached to my laptop with several HID devices attached,  JTAG adapter for flashing, printer etc.

I have only seen a hiccup in the audio streaming a couple times in several months now.  I do see the onboard led come on and stay on sometimes when no audio is streaming, but it goes off immediately when playback starts.