har-in-air/STM32F411_USB_AUDIO_DAC #8: Feature request: volume control

robcazzaro, 2021-05-16:
I really like the code you wrote, and I plan to use it in the future. I'm connecting my STM32F411 BlackPill to a SSM3582 amplifier board that takes I2S input directly and amplifies it (i.e. not just a I2S DAC like the PCM5102a)

Unfortunately, since the amplifier is 30W and has no volume control, without the ability to change the volume, it will always output at the maximum volume. Your code currently only uses volume=0 for mute and volume=100 for full volume, with nothing in between.

Would you consider adding code to actually enable volume control, possibly using some of the ideas in this thread https://www.microchip.com/forums/m932509.aspx?

har-in-air, 2021-05-17:
Thanks, an additional control endpoint for volume is possible but no promises as to when and if I will implement this.
I just use the host application volume control. 

In another project where I use a TAS5753 I2S amplifier, it has an I2C configuration & control interface  with volume and mute control, and I use a rotary encoder for volume control.
If you're talkiing about a similar interface but where I manually scale the I2S data using a button or encoder interface,  that's not something I personally want to add to this project as I prefer  a minimalistic USB-I2S dongle.

har-in-air, 2021-05-17:
BTW, the SSM3582 has an I2C interface, you can add the volume control feature yourself using my STM32F407 USB Audio + FPGA audiocrossover filter + I2S amplifier project as an example

robcazzaro, 2021-05-18:
Thanks. Unfortunately my SSM board does not expose the I2C connection, and doesn't have a volume control. So my only way to change volume is to use the Windows volume control and have Windows send "scaled down" audio. And for some reason it doesn't do it with your code, but works with other USB devices with no volume control.

I think it is connected to how Windows sees the device capabilities. I basically need Windows to recognize the device and uses the following https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/software-volume-control-support

I'm now trying to understand from the USB audio descriptor is there's anything I can do to have Windows behave properly and send scaled down data

Can you actually change volume from your computer? I assume you are running Linux

har-in-air, 2021-05-18:
Yes, I use Ubuntu - VLC, youtube, web video apps with volume controls all work  fine. Only thing that doesn't is the volume control icon on my desktop title bar :-P.

The host apps should see the device enumerated capabilities and either send a control transfer for volume if supported, or scale the data if not supported, like VLC etc are doing. I guess not all support the data scaling option.

robcazzaro, 2021-05-18:
Thanks for confirming, that's what I thought. Even in your case, it's not working properly: if the system volume control doesn't work, that means that the OS thinks that the device will implement its own volume scaling (either in SW or HW, for example with a I2C volume controller). The apps, on the other hand, scale the output internally, before sending it to the OS pipeline. So, obviously, if an app is reducing the volume, the OS itself sees a lower signal and sends a lower signal to the I2S device.

I finally figured out what was happening. In your code, you declared that the USB device has both a Mute and Volume functionality, but then did not implement the volume control itself, just Mute On if Volume==0, or Mute Off for every other value. So basically, every time you change the OS volume, the OS  simply sends a request to the USB endpoint to use a new volume during playback, but then the code didn't do anything with that request.

So the fix is really trivial, simply tell the OS that the device doesn't have a volume control, by removing the |AUDIO_CONTROL_VOL setting in AUDIO_CONTROL_FEATURES as below (line 48 in usb_audio.c)

```
// Feature Unit Config
#define AUDIO_CONTROL_FEATURES (AUDIO_CONTROL_MUTE )	
```

Hardware Mute control in BSP_AUDIO_OUT_SetMute() still works just fine, as well as the HW mute when nothing is playing. On the other hand, BSP_AUDIO_OUT_SetVolume() is never called, so setting the volume to 0 in the OS control will produce no sound (OS scaling the stream to 0), but will not enable the hardware Mute pin. So the only proper way to set a hardware Mute is by using the OS mute functionality

har-in-air, 2021-05-19:
Just removing AUDIO_CONTROL_VOLUME  from AUDIO_CONTROL_FEATURES broke the code (no audio anymore) on my system.

But thanks for pushing this issue, I am working on usb audio mute and volume  control implementation, will update when done.

har-in-air, 2021-05-19:
Implemented USB volume and mute feature. Max volume = 0dB (no amplification), Min volume = -126dB, attenuation steps in 6dB. Now mute and volume control work with VLC as well as Ubuntu desktop volume control. Let me know if you see any issues on Windows.

robcazzaro, 2021-05-19:
I just tested your new code, it works well in Windows, too.

Please keep in mind that on my laptop (Win 7, a few years old), the device keeps flashing the blue LED every time the processor is even a bit more loaded than just background (i.e. window switching). It makes it unusable in practice (every time the blue LED flashes, the audio stream emits a buzz, whose frequency depends on the encoding frequency: 44.1, 48 or 96 KHz)

The same laptop can stream all day to another USB device (Topping MX3) at 96KHz without ever having audible problems. I think that the buffer size you are using might be a bit too small, and too prone to USB data fluctuations. So the STM32 is data starved every time there is an even minor stream problem. Probably Windows is more issue prone than Linux to these problems. I'll see if I can do something about it, and open a new issue with suggestions in that case.

Thanks for all your work and for sharing it

har-in-air, 2021-05-20:
There is a tradeoff with larger buffers - more latency, which becomes an issue with syncing video and audio playback. Anyway, try increasing the buffer size and see if that helps.

closed 2021-05-20 by har-in-air

har-in-air, 2021-05-21:
FYI, updated the volume control feature to use 3dB attenuation steps from 0dB  (max) to -96dB (min)