har-in-air/STM32F411_USB_AUDIO_DAC #9: Minor: the device appears as USB 2.0 capable, when it's only USB 1.1

robcazzaro, 2021-05-18:
This is a common problem with the STM32 USB libraries provided by ST. They define the device as USB 2.0 capable, even when the device itself is only capable of speeds up to 12Mbps, not 480Mbps. In most cases it's harmless, but some USB utilities like USB Tree View (https://www.uwe-sieber.de/usbtreeview_e.html) flag the problem. If the device were really USB 2.0 capable, it would be possible to implement even higher encoding speeds (i.e. 384KHz), but that's not an option with the BlackPill F4x1 which would require a different physical interface for USB OTG HS

In case you want to fix it, it's a simple change in usb_desc.c, lines 89 and 90 (bcdUSB high and low byte)

```
/* USB Standard Device Descriptor */
#if defined ( __ICCARM__ ) /*!< IAR Compiler */
  #pragma data_alignment=4   
#endif
__ALIGN_BEGIN   uint8_t USBD_DeviceDesc[USB_LEN_DEV_DESC] __ALIGN_END = {
  0x12,                       /* bLength */
  USB_DESC_TYPE_DEVICE,       /* bDescriptorType */
  0x10,                       /* bcdUSB */      // was 0x00
  0x01,                                                  // was 0x02
  0x00,                       /* bDeviceClass */
  0x00,                       /* bDeviceSubClass */
  0x00,                       /* bDeviceProtocol */
  USB_MAX_EP0_SIZE,           /* bMaxPacketSize */
  LOBYTE(USBD_VID),           /* idVendor */
  HIBYTE(USBD_VID),           /* idVendor */
  LOBYTE(USBD_PID),           /* idVendor */
  HIBYTE(USBD_PID),           /* idVendor */
  0x00,                       /* bcdDevice rel. 2.00 */
  0x02,
  USBD_IDX_MFC_STR,           /* Index of manufacturer string */
  USBD_IDX_PRODUCT_STR,       /* Index of product string */
  USBD_IDX_SERIAL_STR,        /* Index of serial number string */
  USBD_MAX_NUM_CONFIGURATION  /* bNumConfigurations */
}; /* USB_DeviceDescriptor */
```

har-in-air, 2021-05-19:
Thanks, fixed !

closed 2021-05-19 by har-in-air

har-in-air, 2021-06-10:
Changed back to v2.0, STM32 usb library examples are correct - there is no issue with supporting only FS interface and being compliant with v2.0.