har-in-air/STM32F411_USB_AUDIO_DAC #11: Multiple prototypes (3 units) does works (at sametime) conected in the same hub usb.

FAUSTINOCRUZELIS, 2022-09-14:
hello bro. My name is Fausto and i'm mexican geek of the tecnology and sound systems in general.
I'm trying to develop and little system for power an sound system of 3 ways (bass, middle and high) in my bedroom.
i would like to conect 3 boards (stm32f411... and as i2s-analog bridge i used an pcm5102) in a same device to make easy the setup of my sound system.  To do this i was trying to use some usb hubs , my laptop recognize correctly to these but when i try to use at the sametime only one works and if i try to force to work the others 2 boards, all board go fuck and i must to disconnect the hub to stop the distortion that my action causes (with Carla and pipewire funtion on a fedora linux system), simplify its do not work. I have been thinking that the hub usb cannot give the suficient power to turn all the 3 boards at the sametamie but i made an custom hub usb with an alliexpress' board and external power supply but it did the same behaviour.  now i think that the problem is that the code that you provide (i am very grateful with you and i would like to give you a little reward/payment by your work), gives the same device name and ID (obteined with lsusb to list usb devices and its ports) i think that if i chenge the name and id gived in your code i could use them at the same time correctly. Can you say me the segment from your code were i can change the name and ID (or another parameter to diferential the boards amoung its).

When i was doing research on internet and i found yoy proyect i planteated this (img) where i'm trying to reduce cost and parts. 
[image]

I use it as part of an improviced digital sound proccesor on carla (falkXstudio's linux software) and then crossover the signals to the dacs usb. (img of my segond brobe after try with a comercial hub usb) 
[image]

Greetings and blessings for you.
PD: do you know and alternative to use the stm32f411... as a ADC input for conect an microphone to pc by usb?

[image]

har-in-air, 2022-09-14:
On Windows, the device instance ID is dependent on the descriptor bytes as well as the port it is connected to, so each device should have a different instance ID when they are  plugged in at the same time.
is the USB Instance ID on Windows unique for a device?

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
is the USB Instance ID on Windows unique for a device?

I'm trying to identify a USB device uniquely on a Windows system. Since serial number is optional (and empty on ...
 |

 |

 |

FAUSTINOCRUZELIS, 2022-09-16:
hello i didnt probe this on windows, i only want to make it work on linux.
but i want to set an unique USB Instance ID to each board but i dont know how to do it.

image below is the usb and audio devices listed on device manager on win11.  but with lsusb on windows, the info provided by linux is that apparently the 3 boards have the same id and name (and when i try to use all. )
[image]

har-in-air, 2022-09-16:
If you only want to change the serial number, that is in inc/usb_desc.h - you can modify ID3.
You can also try changing the PID ( I used a random value for PID, the vendor ID is one reserved for prototypes) in  src/usbd_desc.c

FAUSTINOCRUZELIS, 2022-09-17:
i could to change the ids of each board but when i am trying to connecte all at the same hub usb nothing work, only lsubs list them but it linux (pipewire) 
[image]
[image]

but it can't to play audio, it only works when each board is connected to a dedicate usb (1 usb port from laptop for 1 usb dac board) and don't  works connected into the same (bougth on a shop) usb hub  
[image]
the image up is went 2 boards are plugged on diferent usb ports, and the two works normaly.

other caracterist is that, when i plug only one boarn into the hub usb board i works, and then when i plug other board into the same hub, lsusb list both of them correctly but only the first board that was plug, works and the other board don't. 

Thanks bro for your aswers, whenever i can, i going to write an little doc to resume how to implement multiple usb dacs on the same hub usb

FAUSTINOCRUZELIS, 2022-11-05:
I concluded that the issue is in the way what  USB works. And i don't know If the code implementents the USB audio compilant 2, the issue could be sorted.

closed 2022-11-05 by FAUSTINOCRUZELIS