har-in-air/STM32F411_USB_AUDIO_DAC #14: Playing does not start reliably

elagil, 2023-03-14:
Hello,

firstly, thanks a lot for your project. It fixes lots of issues with the ST-provided reference implementation.

I ported the functionality of your code into a fresh CubeMX project and got it playing correctly - the SOF mechanism works great.
However, playing does not start reliably, after stopping previously.

I noticed that - when that happens - I run into the `USBD_AUDIO_IsoOutIncomplete` callback. Then, `USBD_AUDIO_DataOut` is never reached again.

It seems to me like, when hitting `USBD_AUDIO_IsoOutIncomplete`, a previous receive failed. So, I assume that I need to `USBD_LL_PrepareReceive` again in that case.

My solution looks as follows:

```C
static uint8_t USBD_AUDIO_IsoOutIncomplete(USBD_HandleTypeDef *pdev, uint8_t epnum)
{
  UNUSED(epnum);

  USBD_AUDIO_HandleTypeDef *haudio;
  haudio = (USBD_AUDIO_HandleTypeDef *)pdev->pClassDataCmsit[pdev->classId];
  
  USBD_LL_FlushEP(pdev, AUDIO_OUT_EP);

  /* Prepare Out endpoint to receive next audio packet */
  (void)USBD_LL_PrepareReceive(pdev, AUDIOOutEpAdd,
                                &haudio->buffer[haudio->wr_ptr],
                                AUDIO_OUT_PACKET);

  return (uint8_t)USBD_OK;
}
```

Have you encountered similar issues? The above fixes it for me. I never have problems restarting audio playback.

har-in-air, 2023-03-15:
Thanks, that's interesting. I'm not sure I've encountered this and I use the USBDAC as an every day headphone driver - it's plugged in right now.Are you saying you used CubeMX to install  the required libraries as well ? I based mine on the DragonMan Makefile based project and made sure once I had a working set of libraries that they were untouched. That's the reason forstaying with the Makefile based project and a non-standard directory structure.  
But if you can get it working with CubeIDE and CubeMX you should publish it and send me a link that I can add to the README. I'm sure a lot of people would prefer using an up to date CubeIDE/CubeMX build environment.  
Its been a while since I had a working STM development environment on my laptop.  I will update my code with your IsOutIncomplete callback  once I can build and test it. In the meantime if you or anyone else can verify this is compatible with the Makefile project source, let me know.

elagil, 2023-03-15:
> Are you saying you used CubeMX to install  the required libraries as well ?

Yes, that's right! I only edited the minimum amount of files for enabling audio playback. In fact, I only had to add the `usbd_audio` module and kept the rest of the generated files for a USB Audio device.

However, I currently have a reduced feature set, because I don't need the extended functionality (only 16 bit/48 kHz, no volume control), as my DSP/amplifiers run at a fixed 48 kHz. I will create and share a template project, when I am finished.

har-in-air, 2023-03-15:
OK. Are you using Windows ? I have only been using the USB DAC on my Android phone and my Ubuntu laptop. Tested it  once on Windows 10 just to check. I will add your IsOutIncomplete callback.

elagil, 2023-03-15:
Yes, I use Windows 11. Could be related to it, since this is essentially a fix for a host issue.

har-in-air, 2023-03-17:
Added suggested fix

closed 2023-03-17 by har-in-air

elagil, 2023-03-27:
After two weeks of struggle, I still failed to get the current ST driver to send out my feedback value to the PC. It might be broken, as forum posts suggest.

I switched to the amazing ChibiOS HAL and RTOS, and implemented the functionality on that basis.
Find it here: https://github.com/elagil/usb-i2s-bridge

Documentation and functionality are not completely finished, but usable.

har-in-air, 2023-03-28:
I was toying with the idea of migrating the project to an STM32CubeIDE project using the latest HAL libraries, and am now glad I stayed with a working snapshot of the library sources at the time.

elagil, 2023-03-28:
I think that was a wise choice! ST also no longer includes any audio demo projects that I could find - so I assumed that they never test for it.