har-in-air/STM32F411_USB_AUDIO_DAC #21: Add a license to the project, please

DeflateAwning, 2023-12-25:
Can you please add a license? I'm hoping to make some changes, but would need confirmation that it's fair-game to modify and use in a project. Thanks! 

MIT License is a good one, which is preferable, but any other open source one would be fine. 

https://choosealicense.com/ is a great resource, if you have trouble picking one!

har-in-air, 2023-12-25:
I've used a snapshot of the STM32 HAL source code, so you can assume whatever licensing is mentioned in the header files under stm32_usb directory.

closed 2023-12-25 by har-in-air

DeflateAwning, 2023-12-25:
The license used on the STM32 HAL code is a proprietary one of their own, which prevents redistribution in a lot of ways. It is barely an open source licence, and appears to only explicitly permit distribution as "a binary run on an STM32 device". 

It would be great if you could add a real open source licence to this project. It might not mean much to you, but it's important for use of this project in general.