har-in-air/STM32F411_USB_AUDIO_DAC #22: Do you think this project could be made to use the internal clocks on the microcontroller?

DeflateAwning, 2023-12-25:
I'm working on a PCB for this project, and I'm trying to make it as small as possible. Any thoughts on the ability to use the internal clock on the STM32F411 instead of the external crystal clocks on the Black Pill dev kit?

har-in-air, 2023-12-27:
I don't have Stm32CubeIDE installed on my laptop right now, you can open the graphical dialog showing clock configuration and see if  the usb peripheral clock can be generated from the internal RC clock.  The internal RC clock in microcontrollers is usually not accurate or stable enough for reliable USB operation.  I don't remember seeing any datasheet  or app note suggesting that it would be OK. If it was possible, they would definitely have made a big deal of it.

DeflateAwning, 2023-12-27:
I don't see an IOC file in the repo, which is normally what allows you to use the GUI to look at clock/pin/peripheral configurations. Is there another way you know of? 

I'm sure you're right though

har-in-air, 2023-12-28:
It's a makefile project that supports two different cpus. With STM32CubeIDE you can  create a new dummy project for the specific 401 or 411 device and then check out the clock configuration tab.

closed 2023-12-28 by har-in-air