har-in-air/STM32F411_USB_AUDIO_DAC #24: No sound on android phone

rudehardware, 2024-06-15:
Hi, I tried on 2phones, both phones recognized the usb device also configured as the picture from your android device but no sound at all. Working great on pc BTW.

har-in-air, 2024-06-16:
maybe your phone does not support hi-res usb audio (24bit 96khz)

har-in-air, 2024-06-16:
I've tested it with Poweramp app on two Samsung phones (Galaxy F62 and Galaxy F34), was last working about 6 months ago. Unfortunately  the PCM5102A module on my prototype boardwas damaged while trying to reseat it in the board socket, so am not able to confirm whether the Android updates since then have broken anything.

rudehardware, 2024-06-16:
im using samsung galaxy A71  android 13 security patch 1 august 23, after few hours figuring out finaly i found a  solution. i installed hiby music and turned on "enable HQ USB audio access" in the setting. dont know why poweramp doesnt work. 
thanks for the great project

closed 2024-06-21 by har-in-air