har-in-air/STM32F411_USB_AUDIO_DAC #25: blue led flashing and distorted sound

hellcat92, 2024-06-24:
Hi,
I am facing a weird issue where the blue led on the stm32 board is flashing rapidly and i am getting a distorted sound with continuous clicking noise. The sound is missing a lot of details and have a lower pitch. I tried on windows 11 and ubuntu 22.04, problem remains the same. I get same results on all sample rates, just the clicking frequency changes (higher on high samplerate).  Can you please take out some time to guide me in troubleshooting?
Thanks

har-in-air, 2024-06-24:
Try checking if the host  recognises the usb audio capabilities and is resampling the audio to  24bit @ 44.1, 48, or 96k (it should be 24 @ 96khz if you have optimised playback).  You can see the commands for checking the sound playback on Ubuntu, don't know what tools you can use on Windows.My PCM5102 module was damaged some months ago, so don't have a working setup myself.

har-in-air, 2024-06-29:
In case you are not using the black pill boards, check the crystal frequencies on your board.

litezxc, 2024-12-22:
how to add 32 bit?