har-in-air/STM32F411_USB_AUDIO_DAC #26: master clock frequency for 4 wire i2s

Jente105, 2024-06-26:
firstly, thanks for your work on this subject.
I was looking to replace the PCM5102a with a better dac. I noticed that the STM32F411xE configuration could provide the masterclock. what would the frequency of the masterclock be at the rated sample frequencies(44.1, 48 and 96 kHz) of your firmware when masterclock generation is enabled?

thanks,

har-in-air, 2024-06-27:
As per reference manual RM0383 for F411, when MCLK is enabled, MCLK = 256 x Fs (page 585)I've added a comment in the Makefile.

har-in-air, 2024-06-27:
My PCM5102A module died a few months ago (maybe ESD damage), so please add a link to your project when done, I might be tempted to try an upgrade myself.

closed 2024-08-21 by har-in-air